pop-os/nvidia-graphics-drivers #199: nvidia-driver-545 breaks 4k / 144hz displays and only allows 60hz

Bonn93, 2023-12-29:
The latest 545 driver breaks 4k 144hz displays, it only shows 60hz as the max option. However, older versions specifically 470 is fine. I can't install an older version as it seems like they're alised to the latest when trying to set `nvidia-driver-535` 

Why is a specific version aliased to the latest? installing 510 -> installs 545 etc

leviport, 2023-12-29:
Our 4k 144Hz Bonobo WS and Serval WS checked out fine at 144Hz. Is this an external display, and if so, what is the make and model? Which NVIDIA GPU do you have?

Old packages get converted to transitional packages to make sure upgrades happen smoothly, and to make sure old and unmaintained versions of the driver don't get left running on too many users' installs. We occasionally keep a previous release around in case there are troubles with the new release, but we didn't do this with 545, so 535 is also a transitional that points to 545.

Bonn93, 2023-12-29:
It's a GTX 1070 with an Acer Predator 31.5 Inch 4K 144Hz. I can confirm the display port connections and cables are fine, it can do it on the older 470 driver. 

With what you said, shouldn't something like `apt-get install nvidia-driver-510` work then?

leviport, 2023-12-29:
510 is also a transitional, so it'll install 545. It "works" as a transitional package, but it's admittedly a bit unintuitive for anyone who's looking for 510. If you do a `apt search nvidia-driver` you can get a full list of what's transitional and what isn't. There are some `nvidia-driver-{$VERSION}-server` packages of previous versions that come from the Ubuntu repos. If you need a previous version, those are often the easiest option. The fact that they're a server version doesn't seem to make a difference.

Bonn93, 2023-12-29:
Okay, that works, I had to go back to `510` to get 144hz, it seems like nvidia driver issues between 520+ -> 545/latest. 

For anyone else that runs into this, I did; 

```
sudo apt-get remove --purge '^nvidia-.*'
sudo apt-get install nvidia-driver-510-server && sudo reboot
```

zndrr, 2024-02-10:
Sorry to necro, but this also still impacts me. Using an LG C1 at 4K-120hz.

I rolled back to 470 in Dec (I think) after 545.29.06 update broke >60hz, and naively thought maybe resolved by now with newer kernels. Not the case...

Do remember the driver install process to be a PITA and didn't have time to waste on it then -- but am going to have to go through it again.

Bonn93, 2024-02-12:
@zndrr add the following line to `/usr/share/X11/xorg.conf.d/10-nvidia.conf` 

`Option "ModeValidation" "NoExtendedGpuCapabilitiesCheck"` 

Should look like:

```
Section "OutputClass"
    Identifier "nvidia"
    MatchDriver "nvidia-drm"
    Driver "nvidia"
    Option "AllowEmptyInitialConfiguration"
    ModulePath "/usr/lib/x86_64-linux-gnu/nvidia/xorg"
    Option "ModeValidation" "NoExtendedGpuCapabilitiesCheck"
EndSection
```

Restart gdm3 or reboot. Happy refresh-rating, fix coming in a future driver.