pop-os/nvidia-graphics-drivers #208: New hibernating bug on older monitors, fix please!

recoil24, 2024-07-05:
With 556.12 update my monitor now doesn't want to sleep properly, when it goes into sleep mode display splashes for a second and keeps staying in a turned mode without any image on it, inbefore the update everything was okay and display slept off properly.

leviport, 2024-07-08:
Where did you obtain version "556.12"? The currently released version is 550.67, the staged update we're currently testing is 555.58, and I'm not seeing a version 556.xx anywhere on [NVIDIA's official download page](https://www.nvidia.com/en-us/drivers/unix/).

recoil24, 2024-07-08:
> Where did you obtain version "556.12"? The currently released version is 550.67, the staged update we're currently testing is 555.58, and I'm not seeing a version 556.xx anywhere on [NVIDIA's official download page](https://www.nvidia.com/en-us/drivers/unix/).

Oh this is a linux repo for nvidia, i am dumb sorry anyways why u would use linux for gaming at the first place? lmfao

closed 2024-07-08 by recoil24