pop-os/nvidia-graphics-drivers #209: 550 and 555 do not cooperate with Wine games

ws909, 2024-07-05:
After updating the latest supported driver on PopOS in Pop_Shop! to 550 a few days ago, games have been practically unplayable on my computer. I am stuck with either two options:

1. Start up the computer, and go straight to opening a game. I will often (approximately every 30-60 seconds) get a "display signal disconnected" flash, as in, flashes of variations of black colour covering the entire screen, completely replacing the actual video signal, for 2-3 seconds. I'll also have screen tearing in the the bottom of the screen while outside of games (so in the normal desktop interface).

2. The usual solution to number 1, has been to open the _Nvidia X Server Settings_ application, and turn on "Force Full Composition Pipeline" (something I have to do every time awakening the computer, since it resets, and it's impossible to save the settings). Unfortunately, toggling this to on, somehow breaks the driver from cooperating with games. The GPU fans spin up when they are sent work, but the speed of execution seems to be extremely low. It seems to be spinning in some sort of lock (but not a deadlock). Starting up games goes from 2 seconds to several minutes (1,5 hours first time with Apex Legends). When games finally open a fullscreen window to begin rendering, each frame takes around 10-30 seconds to render. Game logic is processed at the same speed. Audio still plays well, for as long as the audio thread is fed work from whichever thread is held back by the graphics driver.

Tested with a variety of games, amongst them Apex Legends in Steam, and Battlefront 2 from the EA App (Lutris).
On the other hand, vanilla, native Minecraft is without issues. I am not sure if that is because Minecraft's bundled binaries are native (such as GLFW), or if it's because Minecraft uses OpenGL instead of Vulkan.

I since updated to 555 ([followed these steps](https://chat.pop-os.org/pop-os/pl/go5ogdw3fbrqznc774hginwmuc): `sudo apt-manage add popdev:nvidia-555.58`), in the hopes my issues would go away without too much work. As far as I can tell, the behaviour is identical.

---

Operating System: Linux-x86_64
NVIDIA Driver Version: 555.58.02
NVML Version: 12.555.58.02
Graphics Processor: NVIDIA GeForce RTX 4080 SUPER
Display: Samsung Odyssey G70B (DP-0) (3840x2160), connected to one of the DisplayPort GPU ports

---

As an aside, my keyboard also randomly seems to get disconnected, while playing games, and sometimes even right after starting from hibernation. This happened after the latest 550 driver, too, but I can't possibly imagine a graphics driver update should have an impact on that.

I also noticed that setting "Force Full Composition Pipeline" in 555 resets the UI scaling to 1x. It's been happening with a small set of the Nvidia drivers I've had installed in the last two months. That is probably a separate issue, however.

jacobgkau, 2024-07-10:
> After updating the latest supported driver on PopOS in Pop!_Shop to 550 a few days ago,

There was no 550 update a few days ago. This is from the commit log:

[image]

550.67 was out for several months. The 555 series is the only thing that's been pushed recently.

jacobgkau, 2024-07-10:
Additional info from Mattermost:

- Motherboard: ASUS PRIME Z790-P WIFI
- Apt history:
    - July: [history.log](https://github.com/user-attachments/files/16168970/history.log)
    - End of June: [history.log.1.gz](https://github.com/user-attachments/files/16168971/history.log.1.gz)

The logs don't show any NVIDIA updates until the user manually installed 555 on 7/5. 555 has not yet been released from master staging; meanwhile, 550.67 was released on March 29th, so the timeline doesn't add up for it to be the cause: https://github.com/pop-os/repo-release/pull/328 

The Linux kernel was updated to 6.9.3 on 6/25. The user's going to try booting into an older kernel to see if it behaves better.

ws909, 2024-07-10:
Booted into "oldkern.conf" (Linux 6.8.0-76060800daily20240311-generic), and the behaviour was as much identical as I am able to confirm.

ws909, 2024-07-11:
I updated my motherboard's UEFI to 1661 (ASUS PRIME Z790-P WIFI), and that immediately improved the situation. I was able to start up Apex Legends, and played 2 or 3 games of the Control gamemode. During this time, the flashes described in point 1 were common, although with longer and less predictable intervals. Occasionally, the monitor would lose the video signal, reporting that there was an error. Replugging the DP cable let the monitor receive a signal again. Switching between Apex Legends and the desktop, caused a lot of issues, most notably a very long loop of 5-20 black screen flashes, 5-infinite seconds video signal loss, and a rare second of video from the game. The last time I replugged the cable, I was stuck with a low framerate, and could not change this until I quit the game (could not stop the game from claiming the screen).

After the last time I replugged the DP cable, I noticed the screen tearing was back, so I enabled "Force Composition Pipeline". I then relaunched Apex Legends. The game launched as fast as it should (no longer hang at "processing shaders" like before the UEFI update), however, the fullscreen game splash screen is rendering at a very low framerate. After 10 minutes, only a few frames have rendered, and the splash screen does not even seem halfway through.

ws909, 2024-07-13:
Managed to test Apex Legends again today in fullscreen, borderless fullscreen, and windowed mode (full resolution framebuffer, but the window is displayed below the menubar and window title bar). In (borderless) fullscreen, all the same issues persist. In windowed, however, the frame rate is quite much lower (obviously), but there are 0 issues other than that.

Is the issue that the DE and fullscreen application is fighting over control of the GPU, and GPU responding to that with "undefined" (had to) behaviour? No idea why "force composition pipeline" would have an effect on that, though.

ws909, 2024-08-16:
Solved it. I also reported my issue on the Nvidia forums, so I'm going to quote [my own reply there](https://forums.developer.nvidia.com/t/4080s-blackouts-crashes-and-freezes/302993/18?u=ws909):

> Yeah, I've located the offending settings. I may have enabled these by myself some while ago; not sure if they default to on or not. Either way I feel very stupid for spending such a long time figuring something as basic as that out, especially as I may have caused all this by enabling the offending setting in the first place.
>
> - Nvidia X Server Settings
>  - X Screen 0
>    - OpenGL Settings:
>      -  Allow Flipping
>
> If this setting is enabled, I experience the behaviour I've experienced when either  "Allow G-SYNC/G-SYNC Compatible" is enabled (blackouts and video output crash), or when "Force Composition Pipeline" is enabled (system freeze when a Wine game's fullscreen window is key.
>
> I guess it's up to Nvidia, PopOS, or X Server developers to decide if this is a bug.