pop-os/nvidia-graphics-drivers #211: Updated over the pop_shop just now, suddenly the 3rd monitor doesn't work anymore

nimblebimble, 2024-07-24:
I just updated from the Pop_Shop!, and now my previously perfectly working 3rd monitor is showing nothing after the restart.

It's present in the settings, and when I disconnect it, it also disappears, and dutifully reappears when I reconnect it, but still shows no screen. The monitor menu shows no input on the HDMI port my PC uses for it, and goes into standby after a while.
Connecting the other via the other HDMI out cable coming from my PC to this monitor works, and I get an image, and putting that same cable in the other monitor also gets no image. 

The working connection is on the graphic card's HDMI output. The non-working one is on a DVI output, via a DVI->HDMI cable, which worked perfectly before the update.
The third monitor is on a DP-DP connection and also still works.

Summary:
The graphics card is an RTX 2060 Super with following outputs: DP, HDMI, DVI. 
Connected are 
DP-DP cable, a 28 inch Samsung monitor, works
HDMI-HDMI cable, 27 inch HP monitor, works
DVI-HDMI cable, another 27 inch HP monitor. Before this update, it worked without any issues whatsoever. No, it doesn't work anymore.

The problematic update went as follows.

**Update 1: This I updated over sudo apt-get update && upgrade**

Upgrade: cosmic-greeter:amd64 (0.1.0~1719238264~22.04~0ac49fe, 0.1.0~1721776291~22.04~acc30ac), linux-system76:amd64 (6.9.3.76060903.202405300957~1718348209~22.04~7817b67, 6.9.3.76060903.202405300957~1721174657~22.04~abb7c06), cosmic-settings-daemon:amd64 (0.1.0~1720020778~22.04~81a5305, 0.1.0~1721272193~22.04~54700df), cosmic-screenshot:amd64 (0.1.0~1715345923~22.04~a8130eb, 0.1.0~1721052047~22.04~08ac28d), cosmic-comp:amd64 (0.1~1718626793~22.04~c91d3fe, 0.1~1721743951~22.04~4f04313), linux-headers-generic:amd64 (6.9.3.76060903.202405300957~1718348209~22.04~7817b67, 6.9.3.76060903.202405300957~1721174657~22.04~abb7c06), cosmic-app-library:amd64 (0.1.0~1719248094~22.04~14bf01d, 0.1.0~1721832181~22.04~5848972), pop-launcher:amd64 (1.2.3~1719269631~22.04~ab7a71a, 1.2.3~1721349795~22.04~e44d6cd), popsicle:amd64 (1.3.3~1707320116~22.04~0c617c9, 1.3.3~1721773298~22.04~3a87912), cosmic-term:amd64 (0.1.0~1718207407~22.04~7c02b9c, 0.1.0~1721650276~22.04~0652001), cosmic-panel:amd64 (0.1.0~1718382292~22.04~d7a3ff6, 0.1.0~1721837073~22.04~4af477d), cosmic-randr:amd64 (0.1.0~1715171387~22.04~8ad117e, 0.1.0~1721287742~22.04~e214fe9), libpop-desktop-widget:amd64 (0.2.0~1708359963~22.04~a375759, 0.2.0~1721078511~22.04~b665120), linux-headers-6.9.3-76060903-generic:amd64 (6.9.3-76060903.202405300957~1718348209~22.04~7817b67, 6.9.3-76060903.202405300957~1721174657~22.04~abb7c06), linux-headers-6.9.3-76060903:amd64 (6.9.3-76060903.202405300957~1718348209~22.04~7817b67, 6.9.3-76060903.202405300957~1721174657~22.04~abb7c06), cosmic-files:amd64 (0.1.0~1720545781~22.04~a3ea6d5, 0.1.0~1721748812~22.04~1820433), cosmic-icons:amd64 (0.1.0~1716304830~22.04~9aad1ab, 0.1.0~1721234019~22.04~73be037), cosmic-edit:amd64 (0.1.0~1720526902~22.04~d7e4693, 0.1.0~1721650308~22.04~33d6522), cosmic-notifications:amd64 (0.1.0~1716252384~22.04~34f44eb, 0.1.0~1721052308~22.04~2caf2fb), linux-generic:amd64 (6.9.3.76060903.202405300957~1718348209~22.04~7817b67, 6.9.3.76060903.202405300957~1721174657~22.04~abb7c06), libnvidia-egl-wayland1:amd64 (1:1.1.13-2pop1~1707162632~22.04~c5241b5, 1:1.1.13-2pop3~1719848365~22.04~527f347), cosmic-greeter-daemon:amd64 (0.1.0~1719238264~22.04~0ac49fe, 0.1.0~1721776291~22.04~acc30ac), firefox:amd64 (1:128.0~1720545818~22.04~208cda6, 1:128.0.2~1721770160~22.04~4339272), cosmic-applets:amd64 (0.1.0~1720535187~22.04~d9d864f, 0.1.0~1721841274~22.04~bd8a7ec), cosmic-workspaces:amd64 (0.1.0~1718210413~22.04~594ecff, 0.1.0~1721334536~22.04~2de3669), popsicle-gtk:amd64 (1.3.3~1707320116~22.04~0c617c9, 1.3.3~1721773298~22.04~3a87912), cosmic-osd:amd64 (0.1.0~1720020394~22.04~43314eb, 0.1.0~1721052322~22.04~27fc5e5), linux-image-generic:amd64 (6.9.3.76060903.202405300957~1718348209~22.04~7817b67, 6.9.3.76060903.202405300957~1721174657~22.04~abb7c06), cosmic-launcher:amd64 (0.1.0~1718305969~22.04~2c3a2fc, 0.1.0~1721275594~22.04~c188029), cosmic-settings:amd64 (0.1.0~1718721001~22.04~6e6403e, 0.1.0~1721708515~22.04~4908e38), xdg-desktop-portal-cosmic:amd64 (0.1.0pop1~1717086453~22.04~5f38044, 0.1.0pop1~1721226949~22.04~813352e), cosmic-store:amd64 (0.1.0~1718281780~22.04~ba3452a, 0.1.0~1721663818~22.04~5d73d53), linux-modules-6.9.3-76060903-generic:amd64 (6.9.3-76060903.202405300957~1718348209~22.04~7817b67, 6.9.3-76060903.202405300957~1721174657~22.04~abb7c06), linux-image-6.9.3-76060903-generic:amd64 (6.9.3-76060903.202405300957~1718348209~22.04~7817b67, 6.9.3-76060903.202405300957~1721174657~22.04~abb7c06), cosmic-session:amd64 (0.1.0~1719514220~22.04~b4561cf, 0.1.0~1721052030~22.04~577a181), linux-libc-dev:amd64 (6.9.3-76060903.202405300957~1718348209~22.04~7817b67, 6.9.3-76060903.202405300957~1721174657~22.04~abb7c06)

**Afterwards, there was still an update notifier in the Pop_Shop! I then installed these updates through the pop shop. After this update I got a system notification to restart, which I did. Afterwards, the above described problems appeared.**

Install: libnvidia-common-555:amd64 (555.58.02-1pop0~1719932335~22.04~7bdc3fd), libnvidia-fbc1-555:amd64 (555.58.02-1pop0~1719932335~22.04~7bdc3fd), libnvidia-fbc1-555:i386 (555.58.02-1pop0~1719932335~22.04~7bdc3fd), libnvidia-gl-555:amd64 (555.58.02-1pop0~1719932335~22.04~7bdc3fd), libnvidia-gl-555:i386 (555.58.02-1pop0~1719932335~22.04~7bdc3fd), libnvidia-extra-555:amd64 (555.58.02-1pop0~1719932335~22.04~7bdc3fd), nvidia-compute-utils-555:amd64 (555.58.02-1pop0~1719932335~22.04~7bdc3fd), nvidia-dkms-555:amd64 (555.58.02-1pop0~1719932335~22.04~7bdc3fd), nvidia-driver-555:amd64 (555.58.02-1pop0~1719932335~22.04~7bdc3fd), libnvidia-encode-555:amd64 (555.58.02-1pop0~1719932335~22.04~7bdc3fd), libnvidia-encode-555:i386 (555.58.02-1pop0~1719932335~22.04~7bdc3fd), nvidia-utils-555:amd64 (555.58.02-1pop0~1719932335~22.04~7bdc3fd), xserver-xorg-video-nvidia-555:amd64 (555.58.02-1pop0~1719932335~22.04~7bdc3fd), libnvidia-decode-555:amd64 (555.58.02-1pop0~1719932335~22.04~7bdc3fd), libnvidia-decode-555:i386 (555.58.02-1pop0~1719932335~22.04~7bdc3fd), nvidia-kernel-common-555:amd64 (555.58.02-1pop0~1719932335~22.04~7bdc3fd), libnvidia-cfg1-555:amd64 (555.58.02-1pop0~1719932335~22.04~7bdc3fd), nvidia-firmware-555-555.58.02:amd64 (555.58.02-1pop0~1719932335~22.04~7bdc3fd), nvidia-kernel-source-555:amd64 (555.58.02-1pop0~1719932335~22.04~7bdc3fd), libnvidia-compute-555:amd64 (555.58.02-1pop0~1719932335~22.04~7bdc3fd), libnvidia-compute-555:i386 (555.58.02-1pop0~1719932335~22.04~7bdc3fd) Upgrade: libnvidia-common-550:amd64 (550.67-1pop0~1710948064~22.04~b0f6b1b, 555.58.02-1pop0~1719932335~22.04~7bdc3fd), libnvidia-fbc1-550:amd64 (550.67-1pop0~1710948064~22.04~b0f6b1b, 555.58.02-1pop0~1719932335~22.04~7bdc3fd), libnvidia-fbc1-550:i386 (550.67-1pop0~1710948064~22.04~b0f6b1b, 555.58.02-1pop0~1719932335~22.04~7bdc3fd), libnvidia-gl-550:amd64 (550.67-1pop0~1710948064~22.04~b0f6b1b, 555.58.02-1pop0~1719932335~22.04~7bdc3fd), libnvidia-gl-550:i386 (550.67-1pop0~1710948064~22.04~b0f6b1b, 555.58.02-1pop0~1719932335~22.04~7bdc3fd), libnvidia-extra-550:amd64 (550.67-1pop0~1710948064~22.04~b0f6b1b, 555.58.02-1pop0~1719932335~22.04~7bdc3fd), nvidia-compute-utils-550:amd64 (550.67-1pop0~1710948064~22.04~b0f6b1b, 555.58.02-1pop0~1719932335~22.04~7bdc3fd), nvidia-dkms-550:amd64 (550.67-1pop0~1710948064~22.04~b0f6b1b, 555.58.02-1pop0~1719932335~22.04~7bdc3fd), nvidia-driver-525:amd64 (550.67-1pop0~1710948064~22.04~b0f6b1b, 555.58.02-1pop0~1719932335~22.04~7bdc3fd), nvidia-driver-550:amd64 (550.67-1pop0~1710948064~22.04~b0f6b1b, 555.58.02-1pop0~1719932335~22.04~7bdc3fd), libnvidia-encode-550:amd64 (550.67-1pop0~1710948064~22.04~b0f6b1b, 555.58.02-1pop0~1719932335~22.04~7bdc3fd), libnvidia-encode-550:i386 (550.67-1pop0~1710948064~22.04~b0f6b1b, 555.58.02-1pop0~1719932335~22.04~7bdc3fd), nvidia-utils-550:amd64 (550.67-1pop0~1710948064~22.04~b0f6b1b, 555.58.02-1pop0~1719932335~22.04~7bdc3fd), xserver-xorg-video-nvidia-550:amd64 (550.67-1pop0~1710948064~22.04~b0f6b1b, 555.58.02-1pop0~1719932335~22.04~7bdc3fd), libnvidia-decode-550:amd64 (550.67-1pop0~1710948064~22.04~b0f6b1b, 555.58.02-1pop0~1719932335~22.04~7bdc3fd), libnvidia-decode-550:i386 (550.67-1pop0~1710948064~22.04~b0f6b1b, 555.58.02-1pop0~1719932335~22.04~7bdc3fd), nvidia-kernel-common-550:amd64 (550.67-1pop0~1710948064~22.04~b0f6b1b, 555.58.02-1pop0~1719932335~22.04~7bdc3fd), libnvidia-cfg1-550:amd64 (550.67-1pop0~1710948064~22.04~b0f6b1b, 555.58.02-1pop0~1719932335~22.04~7bdc3fd), nvidia-kernel-source-550:amd64 (550.67-1pop0~1710948064~22.04~b0f6b1b, 555.58.02-1pop0~1719932335~22.04~7bdc3fd), libnvidia-compute-550:amd64 (550.67-1pop0~1710948064~22.04~b0f6b1b, 555.58.02-1pop0~1719932335~22.04~7bdc3fd), libnvidia-compute-550:i386 (550.67-1pop0~1710948064~22.04~b0f6b1b, 555.58.02-1pop0~1719932335~22.04~7bdc3fd)

Any ideas on what's happening?

nimblebimble, 2024-07-24:
Here's the output of xrandr --query

```
Screen 0: minimum 16 x 16, current 7680 x 1440, maximum 32767 x 32767
DP-1 connected primary 2560x1440+2560+0 (normal left inverted right x axis y axis) 640mm x 360mm
   2560x1440    143.91*+
   1920x1440    143.90  
   1600x1200    143.89  
   1440x1080    143.80  
   1400x1050    143.89  
   1280x1024    143.79  
   1280x960     143.86  
   1152x864     143.92  
   1024x768     143.87  
   800x600      143.83  
   640x480      143.85  
   320x240      142.05  
   1920x1200    143.89  
   1680x1050    143.88  
   1440x900     143.86  
   1280x800     143.84  
   1152x720     143.77  
   960x600      143.72  
   928x580      143.50  
   800x500      143.68  
   768x480      143.69  
   720x480      143.85  
   640x400      143.37  
   320x200      141.40  
   2048x1152    143.88  
   1920x1080    143.88  
   1600x900     143.93  
   1368x768     143.77  
   1280x720     143.67  
   1024x576     143.91  
   864x486      143.63  
   720x400      143.88  
   640x350      143.57  
HDMI-1 connected 2560x1440+5120+0 (normal left inverted right x axis y axis) 600mm x 340mm
   2560x1440     59.91*+
   1920x1440     59.90  
   1600x1200     59.87  
   1440x1080     59.87  
   1400x1050     59.86  
   1280x1024     59.76  
   1280x960      59.94  
   1152x864      59.78  
   1024x768      59.68  
   800x600       59.86  
   640x480       59.38  
   320x240       59.52  
   1920x1200     59.88  
   1680x1050     59.85  
   1440x900      59.89  
   1280x800      59.81  
   1152x720      59.75  
   960x600       59.63  
   928x580       59.88  
   800x500       59.50  
   768x480       59.38  
   720x480       59.71  
   640x400       59.20  
   320x200       58.96  
   2048x1152     59.90  
   1920x1080     59.88  
   1600x900      59.82  
   1368x768      59.88  
   1280x720      59.86  
   1024x576      59.90  
   864x486       59.45  
   720x400       59.55  
   640x350       59.77  
DVI-D-1 connected 2560x1440+0+0 (normal left inverted right x axis y axis) 600mm x 340mm
   2560x1440     59.91*+
   1920x1440     59.90  
   1600x1200     59.87  
   1440x1080     59.87  
   1400x1050     59.86  
   1280x1024     59.76  
   1280x960      59.94  
   1152x864      59.78  
   1024x768      59.68  
   800x600       59.86  
   640x480       59.38  
   320x240       59.52  
   1920x1200     59.88  
   1680x1050     59.85  
   1440x900      59.89  
   1280x800      59.81  
   1152x720      59.75  
   960x600       59.63  
   928x580       59.88  
   800x500       59.50  
   768x480       59.38  
   720x480       59.71  
   640x400       59.20  
   320x200       58.96  
   2048x1152     59.90  
   1920x1080     59.88  
   1600x900      59.82  
   1368x768      59.88  
   1280x720      59.86  
   1024x576      59.90  
   864x486       59.45  
   720x400       59.55  
   640x350       59.77  

```

nimblebimble, 2024-07-24:
If anyone got any ideas or a workaround when you get to this, I would highly appreciate it. Not being able to use one of my monitors is really quite annoying..

nimblebimble, 2024-07-24:
> As a workaround, switching to the 470 driver might work. You can switch between the two versions in the "installed" tab of Pop Shop.

Trying that now

nimblebimble, 2024-07-24:
Yeah, the monitor works again perfectly with 470.239.06.. Trying to switch to the newest one again to see if this changes anything.

nimblebimble, 2024-07-24:
Ok, trying to switch to 555.58.02 over the Pop_Shop! hangs the entire system and leaves one without properly working graphics drivers. Only thing I could do was purge the nvidia stuff over the console, and then reinstall system76-driver-nvidia.
This gets me back to 555.58.02, but still the same problem. Seems 555.58.02 is broken.

nimblebimble, 2024-07-25:
Downgrading to 550 also gets rid of the problem

dkincaid, 2024-07-28:
Thanks for filing this issue. I'm having the same problem here with the 555 driver. It's my second monitor (I only have 2) that stops working after a few minutes, then the whole system freezes. I was able to downgrade to 550 using the commands you provided.

R3G4L101, 2025-12-24:
Same thing with the included 580 driver, my other HDMI monitor is detected in the display settings but does not display anything. Removed the nvidia drivers with commands below. Both monitors are now working with default display drivers, but I need to have an nvidia driver.

I found out that anything above 1080p resolution does not display at all. I can go 4k on the default fallback drivers. So 580 nvidia driver is limited to 1080p on my second monitor/TV.