pop-os/nvidia-graphics-drivers #212: 71-nvidia.rules Missing From CUDA Update

SanBingYouYong, 2024-07-29:
Context: Updated from cuda-11.3 to cuda-12.4, uninstalled with apt-get and installed new version with runfile. 

I followed some instructions (namely [here](https://stackoverflow.com/questions/43022843/nvidia-nvml-driver-library-version-mismatch) and [here](https://askubuntu.com/questions/979259/nvidia-persistence-daemon-continuously-starting-and-stopping-in-syslog)) during the process involving the modification of `71-nvidia.rules` (commenting out the actions related to `nvidia_modeset`). 
 - the issue was with persistence daemon stopping the module unload, so I tried to disable it with what the latter link above suggested but was unsuccessful - only that `nvidia_persistenced.service` was renamed (now renamed back) and `71-nvidia.rules` modified (now missing).

After the installation of 12.4, I figured I might want to change things back including that rules file, but now the rules file is missing. 

I wonder if it has been removed by `sudo apt-get --purge remove cuda-11.3` or the file itself is not in use any more in 12.4 anyways.

mmstick, 2024-07-29:
This file is still being provided by our driver packaging [here](https://github.com/pop-os/nvidia-graphics-drivers/blob/7bdc3fde0bcab1da7ee7fdf2012999e48bc072b1/debian/nvidia-kernel-common-555.install#L9).

This repository does not provide CUDA SDK packages. These are provided upstream by Ubuntu.

SanBingYouYong, 2024-07-29:
Thanks for the swift response. I suppose this is then an issue else where, probably gone with apt-get remove.

closed 2024-07-29 by SanBingYouYong