pop-os/nvidia-graphics-drivers #223: Current version of Nvidia driver is listed in an Nvidia security bulletin

ChaiTRex, 2024-10-31:
`nvidia-smi` reports that my fully updated system is using Nvidia driver version 560.35.03. This version is listed in ["Security Bulletin: NVIDIA GPU Display Driver - October 2024"](https://nvidia.custhelp.com/app/answers/detail/a_id/5586) as being affected.

mmstick, 2024-10-31:
560.35.03 is the latest available driver on their website: https://www.nvidia.com/en-us/drivers/results/
If you're thinking of 565, that is a beta driver, and we don't typically release beta drivers.

ws909, 2024-11-03:
Unless I am misreading, that website lists 560.35.03 as affected, the version currently bundled with PopOS. 565.57.01 resolves the issue. On the other hand, it lists CVEs affected as N/A, thus it seems none of them apply to this version? It's under "Security Updates for NVIDIA Cloud Gaming", and, to the best of my knowledge, PopOS isn't a cloud gaming service, so I'm not sure what I'm looking at.

cgiAlexis, 2024-11-11:
It does look like on Linux desktop R560 is free from this CVE according to the above [link, paragraph linked here](https://nvidia.custhelp.com/app/answers/detail/a_id/5586#:~:text=Affected%20Components%2C%20Affected%20Versions%2C%20and%20Updated%20Versions) 
It appears that the Cloud Gaming entry notes 560.x.x as being affected as noted and doesn't call out 560.x.x for desktop so it's probably fine for now.