pop-os/nvidia-graphics-drivers #226: Suspending when "NVIDIA Graphics" are selected via the Pop_OS switchable graphics would cause a black screen after resuming.

BugsyReportsy, 2025-01-11:
Suspending when NVIDIA Graphics are selected via the Pop_OS switchable graphics would cause the desktop to briefly flash on screen before all screens would turn black, forcing a hard reboot and switching to Hybrid Graphics as a workaround. I noticed this only happened after I either did a clean reinstall of the NVIDIA driver (purge to uninstall first), or after I disabled the nvidia-suspend service. Even after re-enabling nvidia-suspend and then rebooting, the screen would still end up turning black when I try to resume from suspend when NVIDIA Graphics were selected.

BugsyReportsy, 2025-01-14:
To clarify, this seems to only happen when I have an external monitor attached.

BugsyReportsy, 2025-01-14:
Another issue, the screens would randomly blank out after a few minutes of inactivity with NVIDIA Graphics selected, despite screen blanking set to Never in the settings.

BugsyReportsy, 2025-01-15:
The best stop-gap solution I found for now is installing nvidia-driver-550-server, which seems to not have any of the bugs specific to 565.

magicjedi90, 2025-06-28:
> > note that #3442 can't even get to sleep and that you basically have to hard kill the laptop as you can't even change the tty anymore. this is long before waking up.
> > ps I have also tried to disable the Nvidia sleep services as some posted as a solution. it doesn't work either.
> 
> What worked for me was installing nvidia-driver-550-server, then disabling nvidia-suspend and nvidia-hibernate.

this is what got it working for me too - thanks you!