pop-os/nvidia-graphics-drivers #230: System hard lock during bootstrap when GPU code starts loading

AlexisSweetMcKay, 2025-03-21:
"Hard locks during bootstrap, about when the GPU code starts loading. I've been able to reproduce this by installing with the Intel ISO, and later updating to system76-driver-nvidia. The install works, but on boot the system gets to module load and locks. Setting the system to init 3 to avoid Wayland/Xorg also shows the problem. Doing an install from the intel ISO and installing nvidia-driver-470 gives me a working system."

mmstick, 2025-03-21:
What NVIDIA GPU do you have?  Pop 24.04 with COSMIC, or 22.04 with GNOME?

mmstick, 2025-03-21:
Seems NVIDIA released a new 570 driver so I will look into packaging it.

AlexisSweetMcKay, 2025-03-21:
This is on 22.04.
12 GB GeForce RTX 3060 with 3584 CUDA