pop-os/nvidia-graphics-drivers #234: [Pop!_OS 22.04 LTS] GTX 750 TI stuck on 470.256.02, should be 570.133.07

pedropaulosuzuki, 2025-04-14:
Hello everyone! Hope you're doing well.

I'm currently on Pop!_OS 22.04 LTS, and my GPU is the NVIDIA Geforce GTX 750 TI. It should theoreticaly support the latest drivers (as it's a Maxwell GPU), but instead it gets only up to the 470 driver.

[image]

[image]

I don't know if that's the latest driver for Pop!_OS 22.04 LTS or if it's just mistaking the GTX 750 TI as a Kepler GPU instead of Maxwell, so if Pop!_OS is indeed only up to 470 driver, then my bad.

Thanks and best regards,

jacobgkau, 2025-04-14:
Currently, the NVIDIA driver's major version isn't updated automatically (in order to minimize breakage). You should be able to either open the Pop!_Shop and install the 570 driver from the Installed/Drivers page, or install it from the terminal with `sudo apt install nvidia-driver-570`. Do either of those work?

pedropaulosuzuki, 2025-04-15:
> Currently, the NVIDIA driver's major version isn't updated automatically (in order to minimize breakage). You should be able to either open the Pop!_Shop and install the 570 driver from the [...]

It worked! Thanks for the help! Closing the issue.

closed 2025-04-15 by pedropaulosuzuki